bengtmartensson/Infrared4Arduino #5: Support esp8266

probonopd, 2015-12-28:
The esp8266 is a sub-3-EUR module with built-in WLAN and MCU. It can be used to send and receive IR codes over WLAN.

https://github.com/esp8266/Arduino

bengtmartensson, 2015-12-29:
Undoubtly very interesting. I would be very happy for pull requests etc. Note that most of the low-level stuff sits in IRLibTimer, which, at least presently,is almost identical to the same file in IRLib. Unfortunately, I am not that good at the hardware-close details.

bengtmartensson, 2017-04-17:
https://github.com/z3t0/Arduino-IRremote/issues/400

https://github.com/bengtmartensson/Arduino-IRremote/commit/d20a976167ab0e9c00cc8f4f9ac18f5450f844f1

bengtmartensson, 2020-02-22:
After considerable research, I have decided not to support ESP8266 in the sense of

1. Hard PWM sending,  presently class IrSenderPwm,
2. Receiving by equidistant sampling, IrReceiverSampler

However, the board will be supported through the soon-to-be-published hardware independent classes IrSendPwmSoft and its subclasses. Also, it may be possible to implement event-driven reception (like IRremoteESP8266 does).

Rationale: 

1. The board does not support configurable PWM. As opposed to even the simplest ATmega...
2. The board does not support periodic timers with a period < 100 micro seconds (50 micros required). Source: [ESP8266 Non-OS SDK](https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf)  section 3.1.5. (see also 2.3).

Remarks:

1. There is a hack implementing (hardware) PWM by mis-using an UART  by [AnalysIR](https://www.analysir.com/blog/2017/01/29/updated-esp8266-nodemcu-backdoor-upwm-hack-for-ir-signals/) .
2. "But [IRremoteESP8266](https://github.com/crankyoldgit/IRremoteESP8266) supports the board!" Not in the sense above. Sending is with soft pwm, corresponding to my class IrSenderPwmSoftDelay (soon to be pushed); receiving is with interrupt driven reception.
3. Support for ESP32 (#33) is soon to be published.

closed 2020-02-22 by bengtmartensson

probonopd, 2020-03-29:
> However, the board will be supported through the soon-to-be-published hardware independent classes IrSendPwmSoft and its subclasses

Question, is it working on esp8266 now?

bengtmartensson, 2020-03-29:
I just have not published it yet ;-). Planned for tomorrow or possibly today.

Is there any reason why you (seem to) prefer ESP8266m over ESP32?

probonopd, 2020-03-29:
Cool :+1: I just need a way to use an ESP to send and receive codes without the need for a cable, because I need to test some codes for a device which is in another room than the computer... ESP8266 modules are cheaper than ESP32 modules, and I have more of them lying around.

bengtmartensson, 2020-03-30:
> Planned for tomorrow or possibly today.

Sorry, unexpected problems has surfaced. Should be ready RSN however.

probonopd, 2021-07-25:
Are you aware that one can (ab)use the I2S module as a 5-bit PWM generator?

https://shepherdingelectrons.blogspot.com/2018/09/esp8266-minimum-i2s-code.html

> I have now used this I2S technique to `https://shepherdingelectrons.blogspot.com/2018/10/the-ir-egg-voice-controlled-esp8266-i2s.html?view=magazine`

bengtmartensson, 2021-07-25:
> Are you aware that one can (ab)use the I2S module as a 5-bit PWM generator?

Is this the same as the trick by AnalysIR, linked [here](https://github.com/bengtmartensson/Infrared4Arduino/issues/5#issuecomment-589957060) (third "1")?

I will be happy to accept a high-quality PR (basically a src/boards/Esp8266.h, possibly and a *.cpp).

But please note that sending with ESP8266 is already supported using either if the classes IrSenderPwmSoftDelay or IrSenderPwmSpinWait, which is basically what IRremoteESP8266 and IRremote does. It can also be noted that the current IRremote considers "soft-pwm" to be the default...

bengtmartensson, 2025-02-15:
I have decided not to waste my time on ESP8266, because, as I wrote above:

> After considerable research, I have decided not to support ESP8266 in the sense of

>    Hard PWM sending, presently class IrSenderPwm,
>   Receiving by equidistant sampling, IrReceiverSampler

It is not a good board for IR. Period.

However, the ESP32 is supported (sort-of), although I have not touched it for some years. Will do so the next few months.

Unless you think that ~ 1€ for a TSOP xxx38 is too expensive, I also suggest that you do not waste time time on VS* (I included one VS1838 in the bag you got). For entertainment, see https://www.analysir.com/blog/2014/12/08/infrared-receiver-showdown-tsop34438-vs-vs1838b-winner-revealed/

RobThree, 2025-02-15:
> It is not a good board for IR. Period.

Ah, ok. I didn't realize it was "final" 😅

> Unless you think that ~ 1€ for a TSOP xxx38 is too expensive

No, not at all.

> I also suggest that you do not waste time time on VS*

Ah, ok. I'm not familiar with all the parts and there are so many and it's hard to figure out which is better or newer or... I just mentioned it because I thought it would be the interesting part to know about the board. And I just thought the boards might be a cool option for people who aren't that comfortable with soldering.

> (I included one VS1838 in the bag you got).

I know, I have it right in front of me as we speak. Again, thanks for all the parts, much appreciated (I [built mine](https://imgur.com/a/zLwnRj6) as soon as they arrived). I'm just trying to figure out what is what, which is which and what part has advantages over other parts.

>  For entertainment, see https://www.analysir.com/blog/2014/12/08/infrared-receiver-showdown-tsop34438-vs-vs1838b-winner-revealed/

I will. Thanks.

bengtmartensson, 2025-02-16:
Don't get me wrong: The classes in Infrared4Arduino that do not require any particular HW support will run on ESP8266, and is enough to implemente OR sending and receiving. This is described in my previous messages.

Personally, I prefer to spend my efforts on ESP32, which for a very moderate price offers what ESP8266 lacks. And it has a large support through the Homeassistant community.

I find you activities very interesting; please keep me posted.

commit ef8f37d6 (2020-04-02): Support ESP8266, with the exception of hardware PWM. Resolves #5.