bengtmartensson/Infrared4Arduino #30: Help sending unmodulated NEC signal

tenpaiyomi, 2019-06-21:
Hey, first I wanted to say thanks for this library.  It definitely looks like it has taken a lot of work, and I'm super appreciative of people who put in the work and share it with the community.

So, on to the help bit.  I'm trying to take an NEC IR signal:

Address: 9D62
Command: 28
Inverted Command: D7

And send it along as an output to a device, but nothing I seem to do actually seems to work.    Code I used was a modified version of your example Nec1 Renderer

```
#include <Nec1Renderer.h>
#include <IrSenderPwm.h>

const IrSignal *irSignal;

void setup() {
    Serial.begin(115200);
    irSignal = Nec1Renderer::newIrSignal(0x9D62, 0x28, 0xD7);
}

void loop() {
    irSignal->dump(Serial, true);
    IrSenderPwm::getInstance(true)->sendIrSignal(*irSignal, 38);
    delay(10000);
}
```

Now, I'll go ahead and state, I'm not at all familiar with IR or the NEC protocol.  I'm sort of stumbling around trying to make this work, so if there is anything I can do to get some more information and/or help, feel free to let me know and I'll do whatever.

With all that stated, any help you may be able to offer would be greatly appreciated.

bengtmartensson, 2019-06-21:
The NEC1 protocol takes 3 parameters, here called D, S, and F (with a 2-parameter version, S=255-D), each of the parameters between 0 and 255 (inclusive). So your call should possibly be `Nec1Renderer::newIrSignal(0x9D, 0x62, 0x28)` (or .`Nec1Renderer::newIrSignal(0x9D, 0x28)` ). See 
the [newIrSignal API](https://bengtmartensson.github.io/Infrared4Arduino/classNec1Renderer.html#a2eefe1e9c94e56dfa5d94bc83fe8b9e9). You may like to "play around" with [IrScrutinizer](https://github.com/bengtmartensson/harctoolboxbundle/releases/tag/Version-1.4.3) to get some familiarity with the protocol.

 However, I do not know how you arrived at your numbers; In the Arduino world they often get the bit order wrong (NEC1 uses last-significant-bit first bitorder). Compensating for this possible error instead gives the call `Nec1Renderer::newIrSignal(0xB9, 0x14).`

>    `IrSenderPwm::getInstance(true)->sendIrSignal(*irSignal, 38);`

This sends the signal followed by 37 repeats; I suspect this is not what you intend.

What is the meaning of "unmodulated" in the title? (NEC1 is modulated, and IrSenderPwm sends modulated signals.)

tenpaiyomi, 2019-06-21:
So, to expand a little, I'm trying to control a head unit via the remote control wire.  I have one general device that works to control it, but I want to incorporate the control into an all in one electronic that has both some buttons and the ability to send the control signals.

Utilizing [this code](https://simple-circuit.com/arduino-nec-remote-control-decoder/), I read in the numbers that the current device I have is sending out a wire into the head unit.  The results themselves were fairly consistent, so I took them as being valid.  With that stated, they could be anything but, and I'd need to go grab an IR receiver and remote for the head unit, as from everything I read the head unit itself utilizes the NEC protocol and the commands that it uses along the wire are just unmodulated, inverted versions of the NEC IR protocol. (that is, not modulated to 38kHZ, and the mark as a low bit and space as a high bit)

bengtmartensson, 2019-06-21:
You can use the class IrSenderNonMod to send a non modulated signal. Unfortunately, it does not support inversion presently, so you will have to modify IrSenderNonMod.cpp.

tenpaiyomi, 2019-06-21:
Thanks for the information, I'll see what I can work up for that.  I don't currently have access to a PC IR receiver to utilize IrScrutinizer, unless you know of a way to transfer the signal along from an Arduino and breakout IR receiver over USB?

bengtmartensson, 2019-06-21:
http://www.harctoolbox.org/arduino_nano.html, other Arduino than Nano is of course also possible.

tenpaiyomi, 2019-06-21:
Thanks for the link.  I went through and grabbed the Gir lib and compiled the GirsLight against an Uno, connected on COM 5, IR sensor connected to ground, 5v, and signal connected to pin 8.  Compiled and confirmed that the serial monitor properly reports back with version and modules commands, and confirmed that IrScrutinizer properly recognizes the modules when I connect.

Unfortunately, when I press capture and click anything on the IR remote, I get an error:

```
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at org.harctoolbox.IrpMaster.Cleaner.createCookedData(Cleaner.java:108)
	at org.harctoolbox.IrpMaster.Cleaner.<init>(Cleaner.java:52)
	at org.harctoolbox.IrpMaster.Cleaner.clean(Cleaner.java:35)
	at org.harctoolbox.irscrutinizer.GuiMain.processIr(GuiMain.java:1117)
	at org.harctoolbox.irscrutinizer.GuiMain.startButtonActionPerformed(GuiMain.java:8318)
	at org.harctoolbox.irscrutinizer.GuiMain.access$9000(GuiMain.java:83)
	at org.harctoolbox.irscrutinizer.GuiMain$79.actionPerformed(GuiMain.java:2915)
	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
	at java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
```

The sensor I have is a (VMA317)[https://www.jameco.com/z/VMA317-Velleman-1838-Infrared-37-9-KHz-Receiver-Arduino-Compatible-2-pack-_2255218.html].  I did notice on a thread a few years ago, (here)[https://github.com/bengtmartensson/harctoolboxbundle/issues/92], that you stated IrScrutinizer does not work with demodulating sensors.  Is that still the case?

bengtmartensson, 2019-06-21:
> IrScrutinizer does not work with demodulating sensors

That is not quite right. In config.h ,
define RECEIVE but do not define CAPTURE
define IRRECEIVER_1_PIN to the data pin of the demodulating receiver (8 as per above).

tenpaiyomi, 2019-06-21:
You are amazing.

Starting to get somewhere now.  IrScrutinizer gave me the following:

```
protocol = NEC, device = 185, obc = 20

Pronto Hex:
0000 006D 0022 0000 0154 00A7 0016 003E 0016 0016 0016 0016 0016 003E 0016 003E 0016 003E 0016 0016 0016 003E 0016 0016 0016 003E 0016 003E 0016 0016 0016 0016 0016 0016 0016 003E 0016 0016 0016 0016 0016 0016 0016 003E 0016 0016 0016 003E 0016 0016 0016 0016 0016 0016 0016 003E 0016 003E 0016 0016 0016 003E 0016 0016 0016 003E 0016 003E 0016 003E 0016 0478

Raw:
+8900 -4400 +563 -1644 +563 -563 +563 -563 +563 -1644 +563 -1644 +563 -1644 +563 -563 +563 -1644 +563 -563 +563 -1644 +563 -1644 +563 -563 +563 -563 +563 -563 +563 -1644 +563 -563 +563 -563 +563 -563 +563 -1644 +563 -563 +563 -1644 +563 -563 +563 -563 +563 -563 +563 -1644 +563 -1644 +563 -563 +563 -1644 +563 -563 +563 -1644 +563 -1644 +563 -1644 +563 -30100
```

The IrSignalNonMod example seems to most closely match the Raw values, sans the +/- symbols.  Are they relevant or can they simply be dropped?

And on a slightly tangental note, most of the NEC codes I see are usually in the format similar to 0xFF906F or so.  Any general explanation (or link to relevant documentation, as I enjoy reading over it all) how the above translates to the hex codes, or vice versa?

tenpaiyomi, 2019-06-21:
Some more interesting information.  Since I was able to send a demodulated signal, I hooked up the current device to the Uno and sent its signal into IrScrutinizer.  This is the information I got out.

```
protocol = Gap-539-1639-32?, device = 185, subdevice = 70, obc = 235, misc = B9.46.14.EB

Pronto Hex:
0000 006D 0096 0000 00BD 0144 0144 0002 0002 0144 0144 0002 0002 000A 0002 0144 0144 0144 0002 0006 0144 0144 0002 000A 0144 0144 0144 0006 0002 0144 0002 0002 0144 0002 0002 0002 0002 0144 0144 0144 0002 0006 0144 0006 0002 0144 0002 0006 0144 0006 0002 0144 0002 0002 0144 0144 0144 0144 0144 0144 0002 000E 0109 0006 0002 0144 0144 0144 0144 0006 0002 0144 0144 0144 0144 0144 0144 0002 0002 0144 0144 0002 0002 0144 0144 0144 0144 0144 0144 0144 0109 0002 0002 0144 0144 0144 0144 0144 0002 0002 0006 0006 0002 0002 0109 0144 0144 0144 0144 0144 0144 0002 0002 0144 0109 0002 0002 0144 0144 0144 0002 0006 0144 0144 0002 0002 0144 0078 0144 00BD 0014 003F 0014 0014 0014 0014 0014 003F 0014 003F 0014 003F 0014 0014 0014 003F 0014 0014 0014 003F 0014 003F 0014 0014 0014 0014 0014 0014 0014 003F 0014 0014 0014 0014 0014 0014 0014 003F 0014 0014 0014 003F 0014 0014 0014 0014 0014 0014 0014 003F 0014 003F 0014 0014 0014 003F 0014 0014 0014 003F 0014 003F 0014 003F 0014 00BD 0002 0002 0109 017C 0002 0002 0109 0002 0002 0144 0144 00BD 0014 0078 0002 0002 0109 0002 0002 017C 0109 0006 0002 017C 0002 0002 0002 0002 0002 0002 0002 0006 0109 0144 0002 0002 0109 017C 0109 0144 0109 017C 0002 0002 0109 0144 0109 0002 000A 0002 0002 0144 0109 0006 0002 0144 0002 0002 0109 0144 0002 0002 000E 0002 0109 0002 0002 0144 0109 0144 0109 0144 0109 0144 0109 0144 0109 0144 0109 0002 0002 0144 0002 0002 0109 0144 0109 0144 0109 0002 0002 0144 0109 0144 0109 0002 0002 0144 0109 0002 0002 0144 0109 0144 0109 0144 0109 0144

Raw:
+1250 -9130 +8173 -166 +54 -166 +54 -8173 +8173 -166 +54 -8173 +8173 -8173 +8173 -54 +54 -8173 +8173 -166 +54 -8173 +8173 -8173 +8173 -8173 +8173 -54 +54 -8173 +8173 -8173 +8173 -54 +54 -8173 +54 -54 +8173 -8173 +54 -54 +8173 -54 +54 -8173 +54 -54 +8173 -8173 +8173 -54 +54 -8173 +8173 -8173 +8173 -8173 +54 -54 +8173 -8173 +8173 -54 +54 -8173 +8173 -283 +54 -8173 +54 -54 +54 -54 +8173 -54 +54 -8173 +8173 -9130 +8173 -54 +54 -8173 +8173 -8173 +8173 -54 +54 -8173 +54 -54 +8173 -54 +54 -8173 +8173 -8173 +54 -54 +8173 -8173 +8173 -54 +54 -8173 +8173 -8173 +8173 -54 +54 -8173 +400 -54 +8173 -8173 +54 -166 +54 -166 +8173 -54 +54 -8173 +54 -54 +8173 -166 +54 -8173 +8173 -8173 +8173 -166 +54 -8173 +8173 -54 +54 -8173 +8173 -8173 +8173 -8173 +54 -166 +8173 -8173 +166 -54 +8173 -54 +54 -8173 +54 -54 +166 -54 +166 -54 +8173 -9130 +8173 -8173 +8173 -8173 +8173 -8173 +8173 -8173 +54 -54 +8173 -54 +54 -8173 +8173 -54 +54 -8173 +8173 -8173 +8173 -8173 +8173 -54 +54 -8173 +8173 -8173 +8173 -54 +54 -8173 +8173 -8173 +8173 -8173 +8173 -54 +54 -8173 +8173 -8173 +8173 -54 +54 -8173 +8173 -8173 +8173 -8173 +166 -54 +283 -54 +8173 -8173 +54 -283 +8173 -8173 +8173 -54 +54 -8173 +8173 -8173 +8173 -166 +54 -8173 +54 -166 +8173 -54 +54 -8173 +8173 -8173 +8173 -8173 +8173 -166 +54 -8173 +8173 -9130 +54 -166 +8173 -8173 +8173 -8173 +8173 -54 +54 -8173 +8173 -8173 +8173 -54 +54 -8173 +8173 -54 +54 -8173 +8173 -8173 +8173 -8173 +54 -166 +54 -54 +54 -54 +8173 -8173 +54 -54 +8173 -166 +54 -8173 +8173 -54 +54 -8173 +8173 -8173 +8173 -8173 +8173 -8173 +9130 -5200 +550 -1650
```

What's interesting though is that if I scrutinize the remote and send the signal, it spits out `undecodable signal, ignored` unless I spam the button, at which point it will at times spit out the NEC protocol/device/obc in the last post.

bengtmartensson, 2019-06-21:
> ... sans the +/- symbols. Are they relevant or can they simply be dropped?

The signs are pure decoration, and can be ignored. (True to 99%...)

> Any general explanation (or link to relevant documentation, as I enjoy reading over it all) how the above translates to the hex codes, or vice versa?

In IrScrutinizer, the NEC1 protocol is represented as 

    {38.4k,564}<1,-1|1,-3>(16,-8,D:8,S:8,F:8,~F:8,1,^108m,(16,-4,1,^108m)*)[D:0..255,S:0..255=255-D,F:0..255]

which is used to compute the different forms.  Also there are descriptions like this https://www.sbprojects.net/knowledge/ir/nec.php available.

> Since I was able to send a demodulated signal, I hooked up the current device to the Uno and sent its signal into IrScrutinizer. 

Not sure what you are doing here. Sending a non-modulated signal to a de-modulating is not a sane thing -- ideally the demodulator should just block it, giving no output at all.

tenpaiyomi, 2019-06-21:
> Not sure what you are doing here. Sending a non-modulated signal to a de-modulating is not a sane thing -- ideally the demodulator should just block it, giving no output at all.

General thought was to take the device that is sending out an NEC signal over a wire and hook it up to the Uno where the de-modulated signal comes in and see if it picks it up as the same (which, in cases where it is not sending an undecodable signal, it does).

So, the next step I took was to take the D and F values I got from IrScrutinizer and put it into the `Nec1Renderer`, and pass that along to the Uno connected to IrScrutinizer, confirming that I was still getting the expected same protocol information.

So, from here, general thought is:

1. Modify the `Nec1Renderer` class to include a function which simply returns the `IrSequence` for an NEC protocol command.
2. Set up code similar to that defined in the `Nec1Renderer` example to generate a sequence from the new function defined in the `Nec1Renderer` class.
3. Pass that sequence along to `IrSenderNonMod::sendNonModulated` to be output along a ping to the device.

Seem logical?  Or is there a better way you would recommend?

tenpaiyomi, 2019-06-22:
Did some more research based on the information you sent about the NEC protocol.  The following is my findings and results.

The timings result in a volume up command with the following bit stream

```
10011101 01100010 00101000 11010111
9D 62 28 D7
```

Which, of course, you know is the address, inverted address, command, and inverted command.  Knowing that information, and knowing that the NEC protocol precedes with a 9ms burst, 4.5ms space, and followed by a 560us burst, I hard coded the NEC command I got for volume up from IrScrutinizer to just perform a test with it, as follows:

```
#include "Arduino.h"
#include "String.h"

#define SEND_PIN 8U

void sendHighBit() {
  digitalWrite(SEND_PIN, HIGH);
  delayMicroseconds(564);
  digitalWrite(SEND_PIN, LOW);
  delayMicroseconds(1692);
}

void sendLowBit() {
  digitalWrite(SEND_PIN, HIGH);
  delayMicroseconds(564);
  digitalWrite(SEND_PIN, LOW);
  delayMicroseconds(564);
}

void sendVolumeUp() {
  String _bitstream = "";

  // 9ms leading pulse burst
  digitalWrite(SEND_PIN, HIGH);
  delayMicroseconds(9024);

  // Send 4.5ms space
  digitalWrite(SEND_PIN, LOW);
  delayMicroseconds(4512);

  // This is INCREDIBLY POOR.  The sole purpose of this is
  // to currently try brute forcing the command just to see if
  // it can be made to work.

  // Volume up : 
  // Address 10011101
  // Time: 2.25 + 1.12 + 1.12 + 2.25 + 2.25 + 2.25 + 1.12 + 2.25
  // Total: 14.61
  sendHighBit(); _bitstream += "1";
  sendLowBit(); _bitstream += "0";
  sendLowBit(); _bitstream += "0";
  sendHighBit(); _bitstream += "1";
  sendHighBit(); _bitstream += "1";
  sendHighBit(); _bitstream += "1";
  sendLowBit(); _bitstream += "0";
  sendHighBit(); _bitstream += "1";

  // Address Inverted 01100010
  // Time: 1.12 + 2.25 + 2.25 + 1.12 + 1.12 + 1.12 + 2.25 + 1.12
  // Total: 12.35
  sendLowBit(); _bitstream += "0";
  sendHighBit(); _bitstream += "1";
  sendHighBit(); _bitstream += "1";
  sendLowBit(); _bitstream += "0";
  sendLowBit(); _bitstream += "0";
  sendLowBit(); _bitstream += "0";
  sendHighBit(); _bitstream += "1";
  sendLowBit(); _bitstream += "0";

  // Command 00101000
  // Time: 1.12 + 1.12 + 2.25 + 1.12 + 2.25 + 1.12 + 1.12 + 1.12
  // Total: 11.22
  sendLowBit(); _bitstream += "0";
  sendLowBit(); _bitstream += "0";
  sendHighBit(); _bitstream += "1";
  sendLowBit(); _bitstream += "0";
  sendHighBit(); _bitstream += "1";
  sendLowBit(); _bitstream += "0";
  sendLowBit(); _bitstream += "0";
  sendLowBit(); _bitstream += "0";
  
  // Command Inverted 11010111
  // Time: 2.25 + 2.25 + 1.12 + 2.25 + 1.12 + 2.25 + 2.25 + 2.25
  // Total: 15.75
  sendHighBit(); _bitstream += "1";
  sendHighBit(); _bitstream += "1";
  sendLowBit(); _bitstream += "0";
  sendHighBit(); _bitstream += "1";
  sendLowBit(); _bitstream += "0";
  sendHighBit(); _bitstream += "1";
  sendHighBit(); _bitstream += "1";
  sendHighBit(); _bitstream += "1";

  // Final 562.5us burst to signify end of signal
  digitalWrite(SEND_PIN, HIGH);
  delayMicroseconds(564);
  digitalWrite(SEND_PIN, LOW);

  // Total time: 9 + 4.5 + 14.61 + 12.35 + 11.22 + 15.75 = 67.43
  Serial.println("Send Bitstream: " + _bitstream);

  delay(40);
}

void setup() {
  Serial.begin(9600);
  Serial.println("Start");

  pinMode(SEND_PIN, OUTPUT);
}

void loop() {
  sendVolumeUp();
  sendVolumeUp();
  sendVolumeUp();
  delay(10000);
}
```

Verified the information based on the protocol, resulting in a total command of 67.43ms (another page stated the command length is 67.5ms, so I figured a variance of 0.07ms is acceptable).  I also noted that several other implementations of NEC protocol transmitters seem to send the command 2-3 times, so I set mine up to send it 3 times with a 40ms delay between each transmission, then a 10s delay and another burst.

Unfortunately...it doesn't seem to work still.  I'm at a bit of a loss, but am not giving up yet.  If you've any information you might be able to provide here, it would be greatly appreciated.

tenpaiyomi, 2019-06-22:
OK, I'm sort of beating my head against a wall here at this point.  I've not been able to get this to work yet, not sure what I am missing.  I *know* that this functionality is possible, given the various aftermarket products available for controlling a head unit, so I'm not trying the impossible.

bengtmartensson, 2019-06-22:
This is no longer support for Infrared4Arduino, so I suggest that you go to (e.g.) the [Arduino forum](https://forum.arduino.cc). By all means, post a link here. Some comments though,

Possibly you just need to invert the signal, (almost?) all IR demodulating receivers have active low out.

> ... Arduino is going to be 5v based, while head unit operates at 12v

Google for max232

> Modify the Nec1Renderer class to include a function which simply returns the IrSequence for an NEC protocol command.

What is wrong with Nec1Renderer::newIrSignal(...)?

tenpaiyomi, 2019-06-22:
With inverting the signal, do you mean a literal flip of the entire signal? That is, all the way from the initial positive 9ms being a negative 9ms to the address, inverted address becoming inverted address, address, etc?

I'll check out the max 232. I was also going to check out a simple optoisolator component for this as well. 

For the Nec1Renderer, it returned an IrSignal, while all the non own functions were expecting an IrSequence.

bengtmartensson, 2019-06-22:
> With inverting the signal, do you mean a literal flip of the entire signal?

Since postive/negative is a low-level property, only change on the lowest level, Here a smart (too smart?) way

    const bool invert = true;

then change all digitalWrite(SEND_PIN, data) to     digitalWrite(SEND_PIN, data ^ invert);

> For the Nec1Renderer, it returned an IrSignal, while all the non pwm functions were expecting an IrSequence.

IrSignal.getIntro()

tenpaiyomi, 2019-06-22:
I figured it out!  It's a super embarrassing issue that is the cause, and I'll have to adjust my circuit to account for it, but I figured it out!  I didn't have a common ground between the head unit and the circuit, so while it was sending a signal there was no reference to the head unit as to what low was.

Thank you SO much for your help!

closed 2019-06-22 by tenpaiyomi

bengtmartensson, 2019-06-22:
Always stay grounded ;-) !!

Nice that I was able to help.

Bengt