bengtmartensson/Infrared4Arduino #38: Arduino UNO doesn't work

kungccm, 2020-01-14:
Hi Sir,

I try to use ProntoReadDump sample.
I found it is hard to receive the IR signal.
I uncomment "receiver->dump(Serial)".
There is output. However, it seems not completed. So that, it can't decode to Pronto format.
Is there any one has this issue?
Please advise. Thanks.
Output Ex.

> 18:54:35.526 -> +3350 -1850 +350 -500 +400 -1400 +350 -500 +350 -500 +400 -500 +350 -500 +350 -500 +350 -550 +350 -500 +350 -500 +400 -500 +300 -550 +350 -500 +400 -1350 +400 -500 +350 -550 +350 -450 +400 -500 +350 -500 +350 -500 +400 -500 +350 -500 +350 -500 +400 -1400 +350 -500 +350 -550 +300 -500 +400 -500 +400 -450 +350 -500 +400 -500 +350 -500 +350 -500 +400 -500 +350 -1400 +400 -1350 +400 -500 +350 -500 +350 -500 +400 -500 +350 -500 +350 -550 +350 -1400 +350 -1400 +350 -500 +400 -500 +350 -500 +350 -1400 +350 -65535 +3350 -1850 +400 -500 +350 -1450 +350 -450 +350 -550 +350 -500 +400 -450 +350 -550 +350 -500 +350 -500 +400 -500 +350 -500 +350 -500 +400 -500 +350 -1400 +350 -500 +350 -500 +400 -500 +350 -500 +350 -500 +400 -500 +350 -500 +400 -450 +350 -550 +350 -1450 +300 -550 +350 -500 +350 -500 +400 -450 +400 -500 +350 -500 +350 -500 +350 -550 +350 -500 +350 -500 +400 -1400 +350 -1400 +350 -500 +350 -500 +400 -500 +350 -550 +300 -500 +350 -550 +350 -1400 +400 -1400 +300 -550 +350 **-500** 

>

bengtmartensson, 2020-01-14:
I do not see the problem. You uncommented a debugging statement. the output of that looks sort-of OK, (but awfully dirty).  

What do you expect? Show the full sketch.

kungccm, 2020-01-14:
Hi Sir,

The lib version is 1.0.6.
I just modified the RECEIVE_PIN to 8.(I also tried another pin. The result is the same.)
BTW, I just used the same code on MEGA 2560. It work well. 

> 

// This sketch demonstrates the IrReceiverSampler together with Pronto::toHexString.
// It requires a demodulating sensor connected to pin RECEIVE_PIN.
// For this reason, we "guess" the modulation frequency to be
// IrSignal::defaultFrequency.

#include <Arduino.h>
#include <IrReceiverSampler.h>
#include <Pronto.h>

#define RECEIVE_PIN 8U
#define BUFFERSIZE 200U
#define BAUD 115200

IrReceiver *receiver;

void setup() {
    Serial.begin(BAUD);
    receiver = IrReceiverSampler::newIrReceiverSampler(BUFFERSIZE, RECEIVE_PIN);
    receiver->setEndingTimeout(100);
}

void loop() {
    receiver->receive(); // combines enable, loop, disable

    if (receiver->isEmpty())
        Serial.println(F("timeout"));
    else {
        receiver->dump(Serial);
        IrSequence* irSequence = receiver->toIrSequence();
        const char* hex = Pronto::toProntoHex(*irSequence);
        Serial.println(hex);
        delete[] hex;
    }

 }

>

kungccm, 2020-01-14:
Hi Sir,

Below is the output of MEGA 2560:

> 19:34:30.944 -> +3300 -1850 +400 -500 +350 -1450 +350 -450 +400 -500 +350 -500 +400 -500 +350 -450 +400 -500 +400 -450 +400 -450 +400 -500 +350 -500 +400 -500 +350 -1400 +350 -500 +400 -450 +400 -450 +400 -500 +350 -500 +400 -450 +400 -500 +350 -500 +400 -450 +400 -1400 +350 -500 +400 -450 +400 -500 +350 -500 +350 -500 +400 -450 +400 -500 +350 -500 +400 -1400 +350 -450 +400 -1400 +350 -500 +400 -450 +450 -450 +350 -500 +400 -450 +400 -1400 +350 -500 +350 -1450 +350 -450 +400 -500 +350 -500 +400 -450 +400 -1400 +350 -65535 +3300 -1850 +400 -500 +350 -1400 +400 -450 +400 -450 +450 -450 +400 -450 +400 -450 +400 -500 +500 -300 +450 -450 +400 -500 +350 -500 +400 -450 +400 -1400 +350 -700 +200 -450 +400 -450 +400 -500 +350 -550 +350 -450 +400 -500 +350 -500 +400 -450 +400 -1400 +350 -500 +350 -550 +350 -450 +400 -500 +350 -500 +400 -450 +400 -500 +350 -500 +400 -1400 +350 -500 +350 -1400 +350 -550 +350 -450 +400 -500 +350 -500 +400 -450 +400 -1400 +350 -500 +350 -1400 +400 -450 +400 -500 +350 -500 +400 -450 +400 -1400 +350 -65535> 

> 19:34:31.024 -> 0000 006D 0064 0000 007D 0046 000F 0013 000D 0037 000D 0011 000F 0013 000D 0013 000F 0013 000D 0011 000F 0013 000F 0011 000F 0011 000F 0013 000D 0013 000F 0013 000D 0035 000D 0013 000F 0011 000F 0011 000F 0013 000D 0013 000F 0011 000F 0013 000D 0013 000F 0011 000F 0035 000D 0013 000F 0011 000F 0013 000D 0013 000D 0013 000F 0011 000F 0013 000D 0013 000F 0035 000D 0011 000F 0035 000D 0013 000F 0011 0011 0011 000D 0013 000F 0011 000F 0035 000D 0013 000D 0037 000D 0011 000F 0013 000D 0013 000F 0011 000F 0035 000D 09BA 007D 0046 000F 0013 000D 0035 000F 0011 000F 0011 0011 0011 000F 0011 000F 0011 000F 0013 0013 000B 0011 0011 000F 0013 000D 0013 000F 0011 000F 0035 000D 001B 0008 0011 000F 0011 000F 0013 000D 0015 000D 0011 000F 0013 000D 0013 000F 0011 000F 0035 000D 0013 000D 0015 000D 0011 000F 0013 000D 0013 000F 0011 000F 0013 000D 0013 000F 0035 000D 0013 000D 0035 000D 0015 000D 0011 000F 0013 000D 0013 000F 0011 000F 0035 000D 0013 000D 0035 000F 0011 000F 0013 000D 0013 000F 0011 000F 0035 000D 09BA
>

bengtmartensson, 2020-01-14:
Ok. The Uno runs out of memory. (It has only 2.5k of RAM, the string to be printed is > 1000 chars.)

If you cannot use the Mega, you can use the Uno to produce the raw form, and convert it to Pronto using for example [IrScrutinizer](https://github.com/bengtmartensson/harctoolboxbundle), running on a PC. If you flash [AGirs](https://github.com/bengtmartensson/AGirs) on the Arduino, this can be made completely automatic.

kungccm, 2020-01-14:
Hi Sir,

Understood.
Thanks for your prompt reply.

closed 2020-01-14 by kungccm