bengtmartensson/Infrared4Arduino #46: Support board STM32

bengtmartensson, 2020-02-28:
powerful, cheap, small,...

eudoxos, 2020-06-22:
This could serve as an inspiration: https://github.com/MatejGomboc/IRremote-STM32/blob/master/IRremote_port/Src/irSend.c How difficult could the port? With 70MHz at hand, is software-based sender a viable option performancewise?

bengtmartensson, 2020-06-22:
Thanx for the link. It looks (from a first look) fairly promising, so that it should be possible to port that code. Hard? Not extremely hard if you have previous experience; do you plan to work on it?

> With 70MHz at hand, is software-based sender a viable option performancewise?

Will probably work fine. Do you know the classes IrSenderPwmSoftDelay and IrSenderSpinWait, which does not use any particular hardware resources and therefor should be able to run on any sufficiently fast board.

eudoxos, 2020-06-25:
> Will probably work fine. Do you know the classes IrSenderPwmSoftDelay and IrSenderSpinWait, which does not use any particular hardware resources and therefor should be able to run on any sufficiently fast board.
GIven my limited experience with that particular core (not yet even running as I would like it to), soft-PWM is quite sufficient at this stage, I would say.

izzy84075, 2024-10-21:
I might look into this soon. I've got quite a bit more experience with STM32 than I do with ESP32, and I was actually planning on doing this first, but then the regulator on my spare Blue Pill decided to start smoking for no reason. Input Capture and Hardware PWM should both be pretty easy to do, on almost any STM32 MCUs.