bengtmartensson/Infrared4Arduino #56: The easiest question ever...!

pkferrick, 2020-07-14:
I'm attempting to use the library to decode NEC-protocol transmissions.  I am using an ATMega4809 on a breadboard and I have managed to compile and download Nec1Decoder, but so far it isn't working.

The question:  what physical pin on the '4809 corresponds to 5U?  As in:   #define RECEIVE_PIN 5U    

thanks,
Patrick

bengtmartensson, 2020-07-14:
See https://content.arduino.cc/assets/Pinout-NANOevery_latest.pdf

or is it the "U" that is confusing?

pkferrick, 2020-07-14:
It is most certainly the U, as I see no U on either the diagrams you referenced or the DIP40 version of the chip I am using!   Are we talking about digital pin 5, as you would describe it in Arduino/mega328 terms?Tnx,PatrickSent from my Samsung Galaxy , an AT&T LTE smartphone

  (#56) 

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/bengtmartensson/Infrared4Arduino/issues/56#issuecomment-658321005",
"url": "https://github.com/bengtmartensson/Infrared4Arduino/issues/56#issuecomment-658321005",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

bengtmartensson, 2020-07-14:
OK case solved. "U" is a [literal suffix](https://en.wikipedia.org/wiki/Integer_literal#Affixes).

pkferrick, 2020-07-14:
As bad as a programmer as I am, the first thing that came to my mind was
'unsigned int', believe it or not!   So to be clear, I should connect the
receiver to pin 5, which on the Atmega4809 (40 PDIP) is physical pin 38.  I
think that's where I have it, so maybe there is some other problem.   I'll
get it sooner or later!!!

thanks.
Patrick

pkferrick, 2020-07-14:
One other question-  now I'm getting 'No decode', so apparently a signal is
being received but not understood.   I wanted to print the 'raw'
(undecoded?) with something like decoder.printvalue (along the lines of
what I've been doing with other IR libs) but that isn't an option.  I am
sending with protocol NEC, and I wonder if there's an important difference
between that and Nec1.

Patrick

bengtmartensson, 2020-07-15:
> I wanted to print the 'raw'
> (undecoded?) with something like decoder.printvalue

See example IrReceiveSampler.ino.

>I wonder if there's an important difference between that and Nec1.

Nec1 is a certain repeat pattern -- the "standard one". It is unclear what is meant by "NEC".

closed 2020-07-18 by bengtmartensson