bengtmartensson/Infrared4Arduino #57: pinMode, readDigital/writeDigital stops working when #include <IrReceiverSampler.h>

Carsstengithub, 2020-07-15:
Trying the blink example, but when <IrReceiverSampler.h> is included in the standard blink example it stops working on a 
ESP32 Dev Board (Blinking PIN 2), other pins also stops working. But the example IrReceiverSampler works fine, but then I'm unable to use other pins for controlling a relay etc.

Has this something to do with #54 ?

bengtmartensson, 2020-07-15:
Please post a full sketch. 

> Has this something to do with #54 ?

Unlikely; if that bites, compilation will fail.

Carsstengithub, 2020-07-15:
Fulle sketch:
//If this is included it does not work
#include <IrReceiverSampler.h>

#define PIN 2

void setup() {
  pinMode(PIN, OUTPUT);
}

void loop() {
  digitalWrite(PIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(PIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}

AVDeveloppement, 2020-07-30:
Hello, i have same problem with a esp32 can't use a relay, pin don't change
with Arduino nano it work my relay are triggered.
Strange

shawnchu, 2021-08-10:
enum OUTPUT value is 1 in PinModeStatus.h but 2 for ESP:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-gpio.h#L45

The RISING, FALLING enums changed orignal value too.

The best option might be moving the definition to Esp32.h ?

bengtmartensson, 2021-08-11:
I pushed a proposed fix to branch PinModeStatus. Only change is to [PinModeStatus.h](https://github.com/bengtmartensson/Infrared4Arduino/blob/PinModeStatus/src/PinModeStatus.h). Please test.

closed 2021-08-22 by bengtmartensson

shawnchu, 2021-08-24:
Thanks & Verified!

AVDeveloppement, 2021-08-24:
Thanks for your work

commit 31001ebd (2021-08-22): Merge branch 'PinModeStatus'. Fixes #57.