bengtmartensson/Infrared4Arduino #58: Incorrect library.properties architectures entry

majenkotech, 2020-10-19:
In `library.properties` you have both a list of specific architectures *and* the "all" architecture `*`. This is incorrect.  It should either be a list of discrete architectures *or* the `*` wildcard - never both.

Which is it you intended - all architectures or just `avr,megaavr,samd,sam,esp32` ?

bengtmartensson, 2020-10-19:
There are receiving and sending classes (IrSenderPwmSoft* and IrReceiverPoll) that do not use any special hardware properties and thus run on all architectures. So from that point of view, the "*" is correct. Other classes take advantage of special hardware properties, if present, so that motivates the explicitly mentioned architectures. The library runs "better" on those.

I _did_ check official documents when I wrote it, and to my knowledge, the statement is correct and compliant with the Arduino library standard 1.5. If you are of different opinion, please back up your claims with links to official documents. (Possibly it has changed recently?) I am always willing to learn.

majenkotech, 2020-10-19:
The library.properties file applies to the library as a whole, not individual parts of it. It is used to select if a library is valid for a specific architecture or not. It does not describe how the library works, only *if* it works. It doesn't care if there are optimisations for specific architectures, as long as those optimisations don't interfere with other architectures. The inclusion of `*` in the list basically means it works with everything, and that overrides the list of specific architectures making them irrelevant. 

> architectures - (defaults to *) a comma separated list of architectures supported by the library. If the library doesn’t contain architecture specific code use * to match all architectures.

What they mean there is not "contains optimisations for specific architectures" but "will only compile on certain architectures". If your library will compile on any architecture (i.e., it has fallback code using generic API calls rather than specific register calls for example) then the architecture is `*`. If it doesn't have such fallback code and will fail on other architectures (such as stm32 or PIC32) then you mustn't have `*` but a list of only the architectures it will compile on.

majenkotech, 2020-10-19:
Also the `url=` has a typo: `,html` instead of `.html`. As it happens your web server handles that transparently, but it's messy.

majenkotech, 2020-10-19:
Oh, and another thing: you have listed *all* your header files in `includes=`. That is also wrong. You should only list the actual headers the user needs to include into their sketch. If they really do need to include all those headers then you should re-think your layout...

> includes - (available from Arduino IDE 1.6.10) (optional) a comma separated list of files to be added to the sketch as #include <...> lines. This property is used with the "Include library" command in the Arduino IDE. If the includes property is missing, all the header files (.h) on the root source folder are included.

bengtmartensson, 2020-10-19:
architectures: Please read [this](https://arduino.github.io/arduino-cli/latest/sketch-build-process/#architecture-matching). So, yes, my version is sensible and correct.

> Also the url= has a typo: ,html instead of .html

Ok, thanx. Will fix. 

> you have listed all your header files in includes=. 

No, I have not. Please read more carefully. There are two ways of prohibiting an include file from being exported: either put it in a subdirectory of `src`, or define `include=` to be the includes that _should_ be made public/exported. Yes, this is badly documented, and yes, the Arduino IDE is ... not working as an IT professional would desire...

So, yes, the `includes=` line is correct and exactly what it should be.

majenkotech, 2020-10-19:
> No, I have not. Please read more carefully. There are two ways of prohibiting an include file from being exported: either put it in a subdirectory of src, or define include= to be the includes that should be made public/exported. 

Yes, I understand that fine. So you are saying that to use this library it is expected that the end user enter 26 `#includes` at the top of their sketch? That's the purpose of that `includes=` line - to tell the IDE what to insert in the sketch when "include library" is used.

Nothing at all to do with public/private or exported/not exported. Whatever concepts those may be in any particular context.

bengtmartensson, 2020-10-19:
> So you are saying that to use this library it is expected that the end user enter 26 #includes at the top of their sketch? 

In C/C++ programming you include the includes you need -- standard C++ programming. The classes comes with their own headers -- a standard C++ programming idiom. If (and only if) you use all classes you need to include all headers.  (That is actually not quite true, because some headers include other headers.) See the examples.

I welcome a friendly and constructive discussion. This is not a such.

closed 2020-10-19 by bengtmartensson

majenkotech, 2020-10-19:
Hey, if that's how you've designed your library that's fine. I'm not having a go at you. I have seen a lot of very poorly written libraries for Arduino, and seeing such a long includes= line rang immediate alarm bells. 

You should still fix the architectures though. I've worked around it for now in UECIDE's library packaging system so that if it sees a `*` it will ignore any other entries and only build a single global package for all architectures.