bengtmartensson/Infrared4Arduino #59: Simple sendNec1 jams after 2 repeats

KingJimmy27, 2020-11-27:
I am developping a remote controller for a NAD D7050 amplifier on an Arduino Nano.
I use Nec1Decoder.ino to see the codes from my sender.
I have stripped oppo_cooked.ino from the examples to send the codes 135U, 124U, 155U repeatidly.
After two loops the programme crashes.
If I remove the line"delete irSignal;" its runs until out of ram.
Any advice please.

#include <IrSenderPwm.h>
#include <Nec1Renderer.h>
IrSender* irsend = IrSenderPwm::getInstance(true);
static void sendNec1(unsigned D, unsigned S, unsigned F, unsigned times) {
    const IrSignal* irSignal = Nec1Renderer::newIrSignal(D, S, F);
    irsend->sendIrSignal(*irSignal, times);
    delete irSignal;
}
void setup() {
 }
void loop() 
{ 
    delay(1000);
    sendNec1(135U, 124U, 155U, 1);
}

bengtmartensson, 2020-11-28:
Thank you very much for this. I can reproduce, and confirm that it is a bug.

I hope to be able to publish a bug fix soon.

bengtmartensson, 2020-12-16:
I have checked in a fix that I believe solves your problem.  I kindly ask you to verify that it works, then I will release a new version.

KingJimmy27, 2020-12-18:
Sorry for the delay.
That seems to fix the problem.
I had already been using the raw code generation which works perfectly.
Using an Arduino Nano in Visual Micro:
I generated an Arduno Parametized file for my amplifier codes and they all work without crashes.
I also generated a raw Arduino file for comparison of memory used:
Parametized size=6162 bytes with 324 bytes of RAM
Raw size = 6898 bytes with 324 bytes of RAM
Not much difference in the size as this is only about 20% of available I wonder which is the best format to use?
I gues raw is faster when running and Parametized produces a smaller .ino file.

Thanks for the effort, the Scrutinizer is a great programme for buildng the basis of an application and learning about IR coding.

bengtmartensson, 2020-12-19:
Thank you for your feedback, and also thanks for those nice words.

I will close this issue, since the bug is fixed, and I will also make a new release shortly. Feel free to discuss and ask questions in the new [discussions area](https://github.com/bengtmartensson/Infrared4Arduino/discussions).

> Parametized size=6162 bytes with 324 bytes of RAM
> Raw size = 6898 bytes with 324 bytes of RAM
> Not much difference in the size as this is only about 20% of available

This is not the full truth, since memory is allocated dynamically, for example by newIrSignal, which calls new,

>  I wonder which is the best format to use?
> I gues raw is faster when running and Parametized produces a smaller .ino file.

"Normally", a protocol-parameter form is the preferable one if you have the choice; if not for anything else, it looks nicer ;-).

closed 2020-12-19 by bengtmartensson