bengtmartensson/Infrared4Arduino #70: Teensy 3.0 not detected

DaAwesomeP, 2021-05-28:
Hello,

I am getting the following warning when building for Teensy 3.0:
```
Arduino/libraries/Infrared/src/Board.h:263:2: warning: #warning The present board is either unknown, or does not support (Hardware) PWM or equidistant timer sampling. The classes IrReceiverSampler and IrSenderPwmHard will not be available. [-Wcpp]
 #warning The present board is either unknown, or does not support (Hardware) PWM
```
Arduino IDE 1.8.13

bengtmartensson, 2021-05-29:
Hi @DaAwesomeP ,

I can confirm the above error with Arduino 1.8.13 and Teensyduino Version 1.53. Thank you for reporting.

If you change line 241 of Board.h from

    #elif defined(ARDUINO_TEENSY32)

to 

    #elif defined(ARDUINO_TEENSY32) || defined(ARDUINO_TEENSY30)

it will compile. Please do so. Since my Teensy-collection only contains a 3.1, I would like to ask you to confirm that it runs too. Say, for example, the example IrReceiverSampler_SenderPwm. Thank you.

DaAwesomeP, 2021-05-29:
I'm now getting some warnings:
```
In file included from Arduino/libraries/Infrared/src/Board.h:244:0,
                 from Arduino/libraries/Infrared/src/IrSender.h:22,
                 from Arduino/libraries/Infrared/src/IrSenderPwm.h:21,
                 from /tmp/arduino_build_265535/sketch/mysketch.h:27,
                 frommysketch/mysketch.ino:75:
Arduino/libraries/Infrared/src/boards/Teensy3x.h:61:11: warning: ISO C++ forbids declaration of 'reset' with no type [-fpermissive]
     reset() {
           ^
Arduino/libraries/Infrared/src/boards/Teensy3x.h: In member function 'int Teensy3x::reset()':
Arduino/libraries/Infrared/src/boards/Teensy3x.h:64:5: warning: no return statement in function returning non-void [-Wreturn-type]
     }
     ^
```
Full log:
<details>
  <summary>Click to expand (used spoiler since very long)</summary>
<pre><code>
In file included from Arduino/libraries/Infrared/src/Board.h:244:0,
                 from Arduino/libraries/Infrared/src/IrSender.h:22,
                 from Arduino/libraries/Infrared/src/IrSenderPwm.h:21,
                 from /tmp/arduino_build_265535/sketch/mysketch.h:27,
                 frommysketch/mysketch.ino:75:
Arduino/libraries/Infrared/src/boards/Teensy3x.h:61:11: warning: ISO C++ forbids declaration of 'reset' with no type [-fpermissive]
     reset() {
           ^
Arduino/libraries/Infrared/src/boards/Teensy3x.h: In member function 'int Teensy3x::reset()':
Arduino/libraries/Infrared/src/boards/Teensy3x.h:64:5: warning: no return statement in function returning non-void [-Wreturn-type]
     }
     ^
In file included from Arduino/libraries/Infrared/src/Board.h:244:0,
                 from Arduino/libraries/Infrared/src/IrSender.h:22,
                 from Arduino/libraries/Infrared/src/IrSenderPwm.h:21,
                 from /tmp/arduino_build_265535/sketch/InfraredFirmata.h:27,
                 from /tmp/arduino_build_265535/sketch/InfraredFirmata.cpp:17:
Arduino/libraries/Infrared/src/boards/Teensy3x.h:61:11: warning: ISO C++ forbids declaration of 'reset' with no type [-fpermissive]
     reset() {
           ^
Arduino/libraries/Infrared/src/boards/Teensy3x.h: In member function 'int Teensy3x::reset()':
Arduino/libraries/Infrared/src/boards/Teensy3x.h:64:5: warning: no return statement in function returning non-void [-Wreturn-type]
     }
     ^
In file included from Arduino/libraries/Infrared/src/Board.h:244:0,
                 from Arduino/libraries/Infrared/src/IrReceiver.h:7,
                 from Arduino/libraries/Infrared/src/IrReceiverPoll.h:3,
                 from Arduino/libraries/Infrared/src/IrReceiverPoll.cpp:2:
Arduino/libraries/Infrared/src/boards/Teensy3x.h:61:11: warning: ISO C++ forbids declaration of 'reset' with no type [-fpermissive]
     reset() {
           ^
Arduino/libraries/Infrared/src/boards/Teensy3x.h: In member function 'int Teensy3x::reset()':
Arduino/libraries/Infrared/src/boards/Teensy3x.h:64:5: warning: no return statement in function returning non-void [-Wreturn-type]
     }
     ^
In file included from Arduino/libraries/Infrared/src/Board.h:244:0,
                 from Arduino/libraries/Infrared/src/Board.cpp:18:
Arduino/libraries/Infrared/src/boards/Teensy3x.h:61:11: warning: ISO C++ forbids declaration of 'reset' with no type [-fpermissive]
     reset() {
           ^
Arduino/libraries/Infrared/src/boards/Teensy3x.h: In member function 'int Teensy3x::reset()':
Arduino/libraries/Infrared/src/boards/Teensy3x.h:64:5: warning: no return statement in function returning non-void [-Wreturn-type]
     }
     ^
In file included from Arduino/libraries/Infrared/src/Board.h:244:0,
                 from Arduino/libraries/Infrared/src/IrReceiver.h:7,
                 from Arduino/libraries/Infrared/src/IrReceiverSampler.h:3,
                 from Arduino/libraries/Infrared/src/IrReceiverSampler.cpp:1:
Arduino/libraries/Infrared/src/boards/Teensy3x.h:61:11: warning: ISO C++ forbids declaration of 'reset' with no type [-fpermissive]
     reset() {
           ^
[... 137 lines cut ...]
     reset() {
           ^
Arduino/libraries/Infrared/src/boards/Teensy3x.h: In member function 'int Teensy3x::reset()':
Arduino/libraries/Infrared/src/boards/Teensy3x.h:64:5: warning: no return statement in function returning non-void [-Wreturn-type]
     }
     ^
In file included from Arduino/libraries/Infrared/src/Board.h:244:0,
                 from Arduino/libraries/Infrared/src/Pronto.h:18,
                 from Arduino/libraries/Infrared/src/Pronto.cpp:1:
Arduino/libraries/Infrared/src/boards/Teensy3x.h:61:11: warning: ISO C++ forbids declaration of 'reset' with no type [-fpermissive]
     reset() {
           ^
Arduino/libraries/Infrared/src/boards/Teensy3x.h: In member function 'int Teensy3x::reset()':
Arduino/libraries/Infrared/src/boards/Teensy3x.h:64:5: warning: no return statement in function returning non-void [-Wreturn-type]
     }
     ^
Opening Teensy Loader...
Sketch uses 36116 bytes (27%) of program storage space. Maximum is 131072 bytes.
Global variables use 6216 bytes (37%) of dynamic memory, leaving 10168 bytes for local variables. Maximum is 16384 bytes.
</code></pre>
</details>

But it works! I used pin 5 as the README suggested. Thanks!

Also good coincidence: the device I am controlling is indeed a Yamaha receiver that responds to that remote code. :smile:

closed 2021-05-30 by bengtmartensson

bengtmartensson, 2021-05-30:
Nice to hear. Ok to close? I have checked in the changes (including a fix to the warning) and released as version 1.2.1; should be available in the library manager shortly. 

The warning is harmless (in ancient C/C++, declaring functions without return type was OK (int assumed); now no longer OK) but easy to fix, and the checkin fixes it.

Thank you very much for reporting.

commit 5555c1f6 (2021-05-30): Fix Teensy 3.0 support. Resolves #70.