bengtmartensson/Infrared4Arduino #85: HW TIMER NEVER INIT ERROR - ESP32

nccchirag, 2022-09-22:
_timer_group: timer_isr_callback_remove(244): HW TIMER NEVER INIT ERROR
timeout_

I get the above error when compiling sample sketch.

**Dev Env**
Windows 10 x64 PC
ESP32 Dev Module - ESP32-PICO-KIT
Arduino 2.0.5
IR Receiver - TSMP58000

Also, how can I get the frequency from the IR signal?

Please guide
Thank You

bengtmartensson, 2022-09-25:
> I get the above error when compiling sample sketch.

which one? But right now, I am not really in the position to support ESP32s :-(.

> Also, how can I get the frequency from the IR signal?

[IrSignal::getFrequency()](https://bengtmartensson.github.io/Infrared4Arduino/classIrSignal.html#a6080a375ec59889bd0a327a56e0e8a19) . But only IrWidget and derived classes generates IrSignals with non-trivial frequency measurement.

nccchirag, 2022-09-26:
> 

No problem @bengtmartensson .. thank you so much for the response :)

closed 2022-11-11 by bengtmartensson