bengtmartensson/Infrared4Arduino #86: IrReceiverSampler receive() breaks pwm on pins 3 & 11

westminsterflip, 2022-11-10:
After receiver->receiver() called in loop:
[image]
Before call or without it in loop and same from pins 9 & 10 that are working
[image]

It happens both on the breadboard I threw together for the project and a fresh Micro right out of the package.  I am working off of the MultiDecoder example, and if I just comment out the one line for receiver->receive() the pwm continues normally.

bengtmartensson, 2022-11-11:
I am unsure about the point you are trying to make. In particular, is this a hardware problem, a problem with the core Arduino software, or a problem with Infrared4Arduino. Note that an Arduino is a cheap board, not to be expected to perform complicated, multi-tasking real-time applications.

Since I do not know what to do here, and I seriously doubt that it is a Infrared4Arduino bug, I close the issue.

closed 2022-11-11 by bengtmartensson

westminsterflip, 2022-11-14:
I can't see a way that it isn't an effect of the Infrared library.  Here's a sample sketch to illustrate:

```
#include <IrReceiverSampler.h>
#include <MultiDecoder.h>

static constexpr pin_t RECEIVE_PIN = 5U;
static constexpr size_t BUFFERSIZE = 200U;
static constexpr uint32_t BAUD = 115200UL;

IrReceiver *receiver;

void setup() {
  Serial.begin(BAUD);
  receiver = IrReceiverSampler::newIrReceiverSampler(BUFFERSIZE, RECEIVE_PIN);
  analogwrite(11,100);
  //here pwm signal normal
}

void loop() {
  receiver->receive();
  //now pwm waveform clipped & will not trigger
}
```

In my code I'm running 3 LEDs instead of one but that isn't anywhere near a level of complexity the arduino can't handle.
The fact that PWM is only affected after running receive and that the PWM waveform on pin 6 is unaffected after the receive() call points fairly strongly to the library affecting 3&11 somehow

I thought it might be messing with the timer registers, but the frequency is not changing, just the lower peak getting clipped

bengtmartensson, 2022-11-15:
I suggest that you try with the IRRemote library and/or raise you problem in the Arduino forum. I am not presently willing to dig into it.